autonomousvision/transfuser #113: question about generating histogram feature in 2021 branch

BestBUCKY, 2022-10-12:
hello, i'm reading the code of generating lidar histogram feature,but i have a problem.

In your paper you said that x-aixs range is from -16m to 16m,but in the code it seems to be -32m to 32m

    def splat_points(point_cloud):
        # 256 x 256 grid
        pixels_per_meter = 8
        hist_max_per_pixel = 5
        x_meters_max = 16
        y_meters_max = 32
        xbins = np.linspace(-2*x_meters_max, 2*x_meters_max+1, 2*x_meters_max*pixels_per_meter+1)
        ybins = np.linspace(-y_meters_max, 0, y_meters_max*pixels_per_meter+1)
        hist = np.histogramdd(point_cloud[...,:2], bins=(xbins, ybins))[0]
        hist[hist>hist_max_per_pixel] = hist_max_per_pixel
        overhead_splat = hist/hist_max_per_pixel
        return overhead_splat

xbins is setted to [-32, 33], is it a bug?

by the way, im confused about how lidar sensor works,  in #100 said "then use only the front frames by doing an action repeat of 2 for the model",but i did not find this action in the code,can you tell me how to do it?

Thank you for your attention.

Kait0, 2022-10-12:
First question:
Yes I think this is a "bug" (it's just a parameter set differently so you can also see it as a bug in the paper).
We removed the 2* in the [2022 branch](https://github.com/autonomousvision/transfuser/blob/bef1fcf5622c3ab400d4c1f27059786e7cfecb11/team_code_transfuser/data.py#L446) 

Your second question:
Action repeat in [2021 branch](https://github.com/autonomousvision/transfuser/blob/9d5ebf01f7b5f6e5c02ebe0cba423119c465b388/leaderboard/team_code/transfuser_agent.py#L247)

Action repeat in [2022 branch](https://github.com/autonomousvision/transfuser/blob/bef1fcf5622c3ab400d4c1f27059786e7cfecb11/team_code_transfuser/submission_agent.py#L252)

BestBUCKY, 2022-10-13:
Thanks for your reply.

About the second question,Why the auto_pilot agent is collecting the correct lidar data? The lidar data that collected are all front.

Kait0, 2022-10-13:
Good question. The reason is the subsampling.
The simulator is running at 20 FPS, and we store every 10th frame so store the data at 2 FPS.
Now because 10 is an even number we will only ever store even frames.
Now the CARLA LiDAR is operating at 10 Hz / FPS, so when querying it at 20 FPS (simulator, and all other sensors) we will get half a LiDAR sweep at every timestep.
The LiDAR is oriented in such a way that we will get the a front and a back half.
The front halfs you will get at the even frames and the back halfs at the odd frames.

Because we only store even frames we only have LiDAR data from the front in our dataset.

closed 2022-10-13 by BestBUCKY