autonomousvision/transfuser #117: Ask for help about coordinate system in 2022 branch

BestBUCKY, 2022-10-28:
hello, Thanks to your great work! but, i'm confused about the coordinate system, it doesn't look the same as before.

In the DateSet, I didn't understand how to process ego waypoint, but for the target point, i noticed that it is same as 2021 branch, so for the target point, x is the right, y is the back, is it right?

And in model.py, forward_gru function, before use target point, you inverse the y-axis, so the ego waypoint coordinate is right is x-axis and forward is y-axis?

I print the ego waypoint and target point in dataset, it looks strange

ego_waypoint
[[ 7.07202284e-01 -8.91050550e-04]
 [ 2.61902945e+00 -1.10621933e-03]
 [ 4.58310258e+00 -1.65783004e-03]
 [ 6.57283361e+00 -2.60935626e-03]]
target_point
[-4.93408009e-03 -4.62185846e+01]

Kait0, 2022-10-28:
To understand the coordinate system of data you have to trace it from it's source, because it changes throughout the code and we don't have a documentation on it unfortunately.
So the target point comes initially as GPS location and we use the [LBC heuristic](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/submission_agent.py#L615) to convert to CARLA coordinate scales (I did some experiments on that once it is the most accurate conversion that I could find).

What you end up with there is a coordinate system where x is left and y is front (still in global coordinates of the town).
To convert that to the default CARLA coordinate frame (x front y right) we rotate it by 90°.
We also covert is to the ego cars local coordinate frame using the compass and the gps location of the car (denoised with a particle filter). You can find the code for that [here](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/submission_agent.py#L224).

Before feeding the target point to the model we invert the y axis [here](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/model.py#L619) so the target point that the model expects is in x front, y left coordinates.
(There is no good reason for doing that, this conversion is an artifact of a historically grown code base).

Regarding the target point in the data set it depends where you read it from.

The same conversion from global to local coordinates (x front y right) as above happens during the [data loading](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/data.py#L340). The inversion in the forward pass is the same as during inference.

The target point during data collection gets saved [here](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_autopilot/autopilot.py#L327)
and is saved as it is output from the route planner, so x left, y front in global coordinates (the concept of left, front etc. doesn't really make sense in a global coordinate system but you get the idea).

You would need to be more specific where you got the datapoint you printed from.
In the end the only important thing is that the target point is in a local coordinate system and that it is the same one during training and inference.

BestBUCKY, 2022-10-29:
Thank you for your detailed reply!

i'm printing the data after build dataset using index.
So in the dataset, for the target point, the coordinate is x front, y right，and the origin is the ego car, is it true？and i'm confused about the ego waypoint coordinate, i noticed that you get ego waypoint from the label_raw file instead of measurement file, what is the difference? And what is the ego waypoints coordinate?

and for the predict waypoint, before feeding them to the pid_control, you transform them to the local coordinates(x front, y right?), so what is raw predict waypoints coordinate?

and for the printed data,I made a mistake on the index,,so it is ok.

Thanks for your attention

Kait0, 2022-10-31:
> So in the dataset, for the target point, the coordinate is x front, y right，and the origin is the ego car, is it true？

No the target point saved in the file is in global coordinates meaning the origin is the center of the town.
The coordinate system is something like x left, y front (if you render CARLA from a BEV perspective x is left and y is front/up on your screen).

> and i'm confused about the ego waypoint coordinate, i noticed that you get ego waypoint from the label_raw file instead of measurement file, what is the difference?

In the label_raw files we save the bounding boxes of all vehicles including the one of [the ego vehicle](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_autopilot/data_agent.py#L312).
If I remember correctly we read out the position of the ego car from it's bounding box rather than from the measurement file.
Don't think there is a particular reason for that one could use either.
The ego matrix that we save is provided [by CARLA](https://carla.readthedocs.io/en/0.9.10/python_api/#carlatransform) and therefore contains the position of the vehicle in global (town) with the [default CARLA coordinates](https://carla.readthedocs.io/en/0.9.10/python_api/#carlarotation) x front, y right.

BestBUCKY, 2022-11-01:
Thanks for you reply. I got it.

closed 2022-11-01 by BestBUCKY