autonomousvision/transfuser #118: The submission is always in pending

jdmdj1999, 2022-11-01:
Hello, I submitted the result to the leaderboard, but it was always in pending state. Have you ever encountered this problem?

kashyap7x, 2022-11-01:
Hello! The CARLA leaderboard has limited AWS resources and receive a huge increase in the number of submissions at this time of the year, due to the CARLA challenge deadline (07/11/2022). We have also had submissions that were pending for over a week for last year's CARLA challenge. In the past, the CARLA team has often come up with a solution (such as temporarily increasing the resources or extending the challenge deadline).

The leaderboard availability should improve after the deadline.

jdmdj1999, 2022-11-01:
Thanks a lot!

closed 2022-11-01 by kashyap7x