autonomousvision/transfuser #119: ask for help about the segmentation data

BestBUCKY, 2022-11-02:
hello，i‘m using the autopilot agent you provided to generate dataset, but the segmentation data seems have problem.
this is rgb
[image]
this raw segmentaion
[image]
this processed segmentation
[image]

there are many points be considered as Unlabeled, what should i do?

thanks for your attention

kashyap7x, 2022-11-02:
This is an unreal engine bug related to your GPU hardware, see https://github.com/carla-simulator/carla/issues/4328

It seems to often be fixed when launching the CARLA server with the "-opengl" flag, make sure you are doing this. Further, the issue seems to only affect newer GPUs (30xx series).

BestBUCKY, 2022-11-02:
thanks for your reply.
i have launched the CARLA server with ‘opengl’ flag, and this problem arises.
my GPU is 3080ti   :(

BestBUCKY, 2022-11-03:
This problem seems difficult to solve.
i notice that CARLA Leaderbroad2.0 is using CARLA9.13, which support 30 series GPU,  Will your future work be based on the CARLA Leaderbroad2.0 ?

kashyap7x, 2022-11-03:
Our short-term plan is to continue on Leaderboard 1.0 until we resolve the failures of our current systems on that benchmark. We may move to Leaderboard 2.0 next year.

BestBUCKY, 2022-11-04:
Look forward to your future work！

closed 2022-11-04 by BestBUCKY