autonomousvision/transfuser #122: Questions about training with downloaded data

ggosjw, 2022-11-10:
Hi, thank you for opening source your great work.

I have an issue regarding the training with downloaded dataset. 
I use the command chmod +x download_data.sh   and ./download_data.sh to download your dataset.
The downloaded dataset looks like this 
[image]
And i encounter following ValueError when trying to train:
[image]

So could you give me some suggestions regarding this situation?

Thank you in advance for your kind help!

kashyap7x, 2022-11-10:
Please set the `root_dir` arg to the parent folder of your download, `/home/arc/transfuser/data/`.

ggosjw, 2022-11-10:
Thank you very much!
Thanks again for your great work!

closed 2022-11-10 by ggosjw