autonomousvision/transfuser #123: AttributeError: module tensorflow has no attribute contrib

dt-boringtao, 2022-11-13:
Hi

I got "no module called agents" message then I pip install agents, which was successful. Then we I ran the datagen.sh script, it shows the error **"AttributeError: module tensorflow has no attribute contrib"**. 

Then I tried different version of agents, but no luck. Can anyone helps, Thanks!

Tao

Kait0, 2022-11-13:
Agents is a folder in CARLA and not a pip module.
Uninstall the pip module to avoid path conflicts.
make sure you add these three paths to your pythonpath
```
path/to/CARLA_0.9.10.1/PythonAPI/carla
path/to/CARLA_0.9.10.1/PythonAPI
path/to/CARLA_0.9.10.1/PythonAPI/carla/dist/carla-0.9.10-py3.7-win-amd64.egg
```

`datagen.sh /path/to/CARLA_0.9.10.1 /path/to/transfuser`

will do this automatically if you give it if you properly give it the two arguments.

dt-boringtao, 2022-11-13:
Thanks a lot!

Now I moved forward to the next step. Once I ran the datagen.sh with carla and transfuser path. I got an error showing: **no module called srunner**. I found that if I change the import as **import scenario_runner.srunner,** it will work. however, that would require me to go into all relevant files to modify it. Is there a smart way to do it? either through **bashrc or .vscode/setting.json**?

Thanks
Tao

dt-boringtao, 2022-11-14:
I moving forward. Thanks a lot!

Now I got a new erro msg:

**DATAGEN = int(os.environ.get('DATAGEN'))
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'**

closed 2022-11-14 by dt-boringtao