autonomousvision/transfuser #124: Add argument --agent-config=${TEAM_CONFIG} \

dt-boringtao, 2022-11-14:
In the datagen.sh, there is an argument **--agent-config=${TEAM_CONFIG}** \, but I cannot find in the file where TEAM CONFIG is defined. I looked into the files and found that under **team_code_transfuser**, there is a **config.py** file. I tried to use this path, and the carla window will close automatically everytime and showed the following error msg. 

"LowLevelFatalError [File:Unknown] [Line: 3762] 
Failed to link program [Program V_3AA5F7CCAD2B351344BE53DDA22E8BAEBA78720A_P_ABADFE631118A2CF0719C49D3F936DABF0CE8865]. Current total programs: 269, precompile: 0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=144272 LargeMemoryPoolOffset=275408 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)"

any suggestions? Thanks!

Tao

Kait0, 2022-11-14:
TEAM_CONFIG is a folder that should contain an args.txt file and your model weights.
You can either train a model yourself or download our pre-trained weights.
Instructions are in the readme.

dt-boringtao, 2022-11-14:
Thanks! 

I followed the instruction, then I got an error msg:

"./leaderboard/scripts/local_evaluation.sh: line 32: 60188 Segmentation fault      (core dumped) python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator_local.py --scenarios=${SCENARIOS} --routes=${ROUTES} --repetitions=${REPETITIONS} --track=${CHALLENGE_TRACK_CODENAME} --checkpoint=${CHECKPOINT_ENDPOINT} --agent=${TEAM_AGENT} --agent-config=${TEAM_CONFIG} --debug=${DEBUG_CHALLENGE} --resume=${RESUME}
"

Kait0, 2022-11-15:
That is too generic an error message to say much what is going on. Segmenation faults can appear if CARLA crashes. Does the simulator run properly (in another terminal)?
What command did you use to start the script?
Did you properly give the arguments to the script:
e.g. 
`./leaderboard/scripts/local_evaluation.sh <carla root> <working directory of this repo (*/transfuser/)>`

closed 2022-11-21 by Kait0