autonomousvision/transfuser #126: question about pixels_per_meter

huohuangcanglong, 2022-11-24:
I'm confused about this (https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/config.py#L16).How to obtain this parameter value? I guess it is related to this(https://github.com/carla-simulator/carla/blob/0c41f167cf1b3f33e23e3be6fb5e1c9552ba5969/PythonAPI/examples/lidar_to_camera.py#L120), but I don't know how to obtain it. I'm looking forward to your answer.

Kait0, 2022-11-25:
Pixels per meter describes the width and length of a pixel/voxel in the LiDAR image.
It has nothing to do with the camera, this is a choice/hyperparameter (that is dependant on a few others).

If you want to compute it:
[lidar_resolution_width](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/config.py#L14) / ([max_x](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/config.py#L45) - [min_x](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/config.py#L44)) = [pixels_per_meter](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/config.py#L16)

huohuangcanglong, 2022-11-29:
> Pixels per meter describes the width and length of a pixel/voxel in the LiDAR image. It has nothing to do with the camera, this is a choice/hyperparameter (that is dependant on a few others).
>
> If [...]

Thanks for your reply. I understand what you mean. If I want to know the pixels_per_meter in the rgb image, whether it is equal to image_ with / (2.0 * (fov * 3.14 /  360.0) * D), where D represents the distance between the camera and the object?

Kait0, 2022-11-29:
The camera image is a perspective grid. Each pixel does not necessarily cover the same amount of meters so I don't quite understand the question.
This variable doesn't really make sense for the rgb image.

closed 2022-12-03 by Kait0