autonomousvision/transfuser #127: ask for help about the transfuser agent

BestBUCKY, 2022-11-26:
hello, i have some problem about the difference of the agent between 2021 branch and the 2022 branch
As the network makes a prediction every two steps, for the step of not making predictions, 2021 branch directly uses the predicted waypoints of previous step, but in the 2022 branch, there are some changes. Can you tell me the details about the changes https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/submission_agent.py#L253
Thanks for your attention

Kait0, 2022-11-26:
This update_gps_buffer is for the custom particle filter that we use to denoise the GPS signal.
The filter is running at the full 20 Hz which is why it is called every frame.
It does not have any effect on the repeated trajectory.

The filter is described in section 2.1 of the [appendix](https://www.cvlibs.net/publications/Chitta2022PAMI_supplementary.pdf).

BestBUCKY, 2022-11-28:
thanks for your reply. i will take a look at appendix

closed 2022-11-28 by BestBUCKY