autonomousvision/transfuser #128: problem in evaluation

xanhug, 2022-11-27:
thanks for your amazing work!
when i try to operate the "local_evaluation.sh" as the instruction by
`/local_evaluation.sh /home/hxa/TF/transfuser/carla /home/hxa/TF/transfuser`
i found that nothing is happen except carla's view is changed into here;
there is no ago agent or other vehicles.
[image]
and the code is like this:
`(tfuse) hxa@hxa-Nitro-AN515-58:~/TF/transfuser/leaderboard/scripts$ ./local_evaluation.sh /home/hxa/TF/transfuser/carla /home/hxa/TF/transfuser
/home/hxa/TF/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py:89: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(dist.version) < LooseVersion('0.9.10'):
> Registering the global statistics
(tfuse) hxa@hxa-Nitro-AN515-58:~/TF/transfuser/leaderboard/scripts$ 
`
i wonder if the carla will running like the video https://www.youtube.com/watch?v=DZS-U3-iV0s&list=PL6LvknlY2HlQG3YQ2nMIx7WcnyzgK9meO if my code working normally?
what should i do now to perform the evaluation normally?

i'm sorry for my poor english,hope i could describe my problem clearly.

Kait0, 2022-11-29:
GPU memory should be fine I can also run these models on my old 980 with 4 GB in high quality.

I would suggest you to use a debugger and debug in the leaderboard_evaluator_local.py file to see why it directly skips to the 

> Registering the global statistics

step.
The program should go to [this function](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/leaderboard/leaderboard/leaderboard_evaluator_local.py#L411):

xanhug, 2022-11-30:
> 

thanks for your reply!
i'll try to check it as your suggestion.

xanhug, 2022-12-01:
> GPU memory should be fine I can also run these models on my old 980 with 4 GB in high quality.
>
> I would suggest you to use a debugger and debug in the leaderboard_evaluator_local.py file to see why [...]

i have try to redownload transfuser and follow the readme.md, and this time i running the "local_evaluation.sh", something is different,seems it's also a question about gpu memory?
[image]
[image]

when i training with the dataset(210g) few days before,i had encounter similar question,but i finally training successfully by decrease batch_size to '2'
but now, there seems no batch size could be modify, could you tell me what should i do please?

xanhug, 2022-12-01:
> GPU memory should be fine I can also run these models on my old 980 with 4 GB in high quality.
>
> I would suggest you to use a debugger and debug in the leaderboard_evaluator_local.py file to see why [...]

there are 3 pth files
[image]
and i only keep one in this directory
[image]
now it seems get to next step like that 
[image]
is that right? it seems different from the YOUTUBE video like https://www.youtube.com/watch?v=DZS-U3-iV0s&list=PL6LvknlY2HlQG3YQ2nMIx7WcnyzgK9meO
in this video,there are depth graph, point cloud graph and the perspective is different too. So i don't know if my evaluation program running correctly.

kashyap7x, 2022-12-01:
The CARLA display you get is correct. Removing the 2 pth files makes the code evaluate a single model rather than the ensemble of 3 models which reduces the memory requirement.

We generated the YouTube videos by saving the model inputs and outputs to the disk while evaluating, using this function:
https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/model.py#L935
You can enable the visualization by adding a `SAVE_PATH` environment variable to `local.evaluation.sh`. The visualized frames will be stored in this `SAVE_PATH`. Keep in mind that enabling the visualization will slow down the evaluation.

xanhug, 2022-12-02:
> 

i got it,thanks a lot.

closed 2022-12-02 by xanhug