autonomousvision/transfuser #130: What's the wp loss on town2 and town5

raozhongyu, 2022-12-01:
Thanks for your great job. Recently, I try to make some changes based on the transfuser dataset.  It's too much time to evaluate the performance on carla every time. Hence, I'd like to what's your wp loss  when evaluate on town2 and town5?

Kait0, 2022-12-01:
I would be very careful with waypoint valdation loss.
It is a bad predictor of driving score. (I usually don't even check it because it is quite pointless).

I agree that evaluating on CARLA can be slow if done sequentially.
What we do to speed up evaluation is we evaluate every route in parallel using some scripts specific to our cluster and then aggregate the results using the [results parser](https://github.com/autonomousvision/transfuser/blob/2022/tools/result_parser.py).

In case you have limited ressources I can recommend the [LAV routes](https://github.com/dotchen/LAV) for evaluation. They are fewer routes in total then longest 6 and the traffic density is lower, which speeds up the simulator.

raozhongyu, 2022-12-01:
Thanks  a lot, I get it!

closed 2022-12-03 by Kait0

raozhongyu, 2022-12-08:
I'd like to konw can I get the Extrinsic of the camera from your meansurements?  Is it the same with the ego_matrix?

Kait0, 2022-12-08:
We use three different cameras.
Their configuration can be found [here](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/config.py#L20).
At inference we only use the middle 60° FOV of each camera and then concatenate them together to form one image.

raozhongyu, 2022-12-09:
Thanks a lot.