autonomousvision/transfuser #132: Problems by data generation

JianLiMech, 2022-12-19:
Hello,

thank you very much for your great work! It meant a lot for my work!

But I have a problem while I was trying to run datagen.sh file.

The error was
 ```
(tfuse) jian@jian-GJ5CN64:~/Masterthesis/transfuser/leaderboard/scripts$ ./datagen.sh
python3: can't open file '/leaderboard/leaderboard/leaderboard_evaluator_local.py': [Errno 2] No such file or directory
```
I have already changed the first 2 lines of datagen.sh file. See below:
```
export CARLA_ROOT=${1:-/home/Masterthesis/carla}
export WORK_DIR=${2:-/home/Masterthesis/transfuser}

export CARLA_SERVER=${CARLA_ROOT}/CarlaUE4.sh
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg
export SCENARIO_RUNNER_ROOT=${WORK_DIR}/scenario_runner
export LEADERBOARD_ROOT=${WORK_DIR}/leaderboard
export PYTHONPATH="${CARLA_ROOT}/PythonAPI/carla/":"${SCENARIO_RUNNER_ROOT}":"${LEADERBOARD_ROOT}":${PYTHONPATH}

export SCENARIOS=${WORK_DIR}/leaderboard/data/training/scenarios/Scenario10/Town10HD_Scenario10.json
export ROUTES=${WORK_DIR}/leaderboard/data/training/routes/Scenario10/Town10HD_Scenario10.xml
export REPETITIONS=1
export CHALLENGE_TRACK_CODENAME=MAP
export CHECKPOINT_ENDPOINT=${WORK_DIR}/results/Town10HD_Scenario10.json
export SAVE_PATH=${WORK_DIR}/results/Town10HD_Scenario10
export TEAM_AGENT=${WORK_DIR}/team_code_autopilot/data_agent.py
export DEBUG_CHALLENGE=0
export RESUME=1
export DATAGEN=1

python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator_local.py \
--scenarios=${SCENARIOS}  \
--routes=${ROUTES} \
--repetitions=${REPETITIONS} \
--track=${CHALLENGE_TRACK_CODENAME} \
--checkpoint=${CHECKPOINT_ENDPOINT} \
--agent=${TEAM_AGENT} \
--agent-config=${TEAM_CONFIG} \
--debug=${DEBUG_CHALLENGE} \
--resume=${RESUME}
```
I would be really nice of you if you could have a look on this problem.

Kait0, 2022-12-19:
Sound like the command 
`./datagen.sh`

is still overwriting the work dir variable with an empty string.
Try:
`./datagen.sh /home/Masterthesis/carla /home/Masterthesis/transfuser`

JianLiMech, 2022-12-19:
Thanks for quick reply!

I tried it but still it's not working. The error is the same.
`(tfuse) jian@jian-GJ5CN64:~/Masterthesis/transfuser/leaderboard/scripts$ ./datagen.sh /home/Masterthesis/carla /home/Masterthesis/transfuser
python3: can't open file '/home/Masterthesis/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py': [Errno 2] No such file or directory
`

JianLiMech, 2022-12-19:
I tried to delete this line, and another same error came:
```
derboard_evaluator_local.py': [Errno 2] No such file or directory
./datagen.sh: line 26: --scenarios=/home/Masterthesis/transfuser/leaderboard/data/training/scenarios/Scenario10/Town10HD_Scenario10.json: No such file or directory
```
So I guess the problem is that, the system can't find the PATH I just exported, but I still have no solution.

Kait0, 2022-12-19:
Your first error was missing one leaderboard folder:
> python3: can't open file '/home/Masterthesis/transfuser/leaderboard/leaderboard_evaluator_local.py': [Errno 2] No such file or directory

The second one you posted had the correct path.

> python3: can't open file '/home/Masterthesis/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py': [Errno 2] No such file or directory 

If python can not open that file that means for some reason it is not there. Check on your file system that the file leaderboard_evaluator_local.py exists at the correct path.

The path you exported seems to be working fine, otherwise the code could not have pieced together the full path.

> I tried to delete this line, and another same error came:

There is no point in deleting that line. The script is set up to start this python file.
The backslashes behind the commands mean that all the following lines are one command.
The lines afterwards are options for the program.

closed 2022-12-26 by Kait0