autonomousvision/transfuser #135: Error will local evaluation: the ego vehicle move randomly

LeBronLiHD, 2022-12-23:
dear developers, 

after build the environment by [issue #134](https://github.com/autonomousvision/transfuser/issues/134#issue-1508093018), I am excited to see the transfuser agent's performance, but the ego car **do not move** after I ran the script.

[image]

after a while, when there is `Detected agent being stuck. Move for frame:  XXX` print in terminal, the ego car finally start moving. 
however, after trying many times, the ego car just **move randomly**, and usually drive out of the road and then crash.

[image]

here is the log of one iteration:

```python
(tfuse) haodong@rog:~/Projects/Ali_DAMO_AVTest/carla_apollo/transfuser$ ./leaderboard/scripts/local_evaluation.sh /home/haodong/Projects/Ali_DAMO_AVTest/carla_apollo/transfuser/carla /home/haodong/Projects/Ali_DAMO_AVTest/carla_apollo/transfuser
/home/haodong/Projects/Ali_DAMO_AVTest/carla_apollo/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py:89: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(dist.version) < LooseVersion('0.9.10'):

========= Preparing RouteScenario_26 (repetition 0) =========
> Setting up the agent
/home/haodong/Projects/Ali_DAMO_AVTest/carla_apollo/transfuser/model_ckpt/transfuser/model_seed1_39.pth
> Loading the world
No more spawn points to use. Spawned 303 actors out of 500
> Running the route
/home/haodong/Applications/Miniconda/Miniconda/envs/tfuse/lib/python3.7/site-packages/mmdet/models/utils/gaussian_target.py:227: UserWarning: __floordiv__ is deprecated, and its behavior will change in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor').
  topk_clses = topk_inds // (height * width)
/home/haodong/Applications/Miniconda/Miniconda/envs/tfuse/lib/python3.7/site-packages/mmdet/models/utils/gaussian_target.py:229: UserWarning: __floordiv__ is deprecated, and its behavior will change in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor').
  topk_ys = topk_inds // width
Detected agent being stuck. Move for frame:  0
Detected agent being stuck. Move for frame:  1
Detected agent being stuck. Move for frame:  2
Detected agent being stuck. Move for frame:  3
Detected agent being stuck. Move for frame:  4
Detected agent being stuck. Move for frame:  5
Detected agent being stuck. Move for frame:  6
Detected agent being stuck. Move for frame:  7
Detected agent being stuck. Move for frame:  8
Detected agent being stuck. Move for frame:  9
Detected agent being stuck. Move for frame:  10
Detected agent being stuck. Move for frame:  11
Detected agent being stuck. Move for frame:  12
Detected agent being stuck. Move for frame:  13
Detected agent being stuck. Move for frame:  14
Detected agent being stuck. Move for frame:  0
Detected agent being stuck. Move for frame:  1
Detected agent being stuck. Move for frame:  2
Detected agent being stuck. Move for frame:  3
Detected agent being stuck. Move for frame:  4
Detected agent being stuck. Move for frame:  5
Detected agent being stuck. Move for frame:  6
Detected agent being stuck. Move for frame:  7
Detected agent being stuck. Move for frame:  8
Detected agent being stuck. Move for frame:  9
Detected agent being stuck. Move for frame:  10
Detected agent being stuck. Move for frame:  11
Detected agent being stuck. Move for frame:  12
Detected agent being stuck. Move for frame:  13
Detected agent being stuck. Move for frame:  14
Detected agent being stuck. Move for frame:  0
Detected object directly in front of the vehicle. Stopping. Step: 4518
Detected agent being stuck. Move for frame:  1
Detected object directly in front of the vehicle. Stopping. Step: 4520
Detected agent being stuck. Move for frame:  2
Detected object directly in front of the vehicle. Stopping. Step: 4522
Detected agent being stuck. Move for frame:  3
Detected object directly in front of the vehicle. Stopping. Step: 4524
Detected agent being stuck. Move for frame:  4
Detected object directly in front of the vehicle. Stopping. Step: 4526
Detected agent being stuck. Move for frame:  5
Detected object directly in front of the vehicle. Stopping. Step: 4528
[... 95 lines cut ...]
│ Criterion             │ Result  │ Value   │
├───────────────────────┼─────────┼─────────┤
│ RouteCompletionTest   │ FAILURE │ 0.54 %  │
├───────────────────────┼─────────┼─────────┤
│ OutsideRouteLanesTest │ SUCCESS │ 0 %     │
├───────────────────────┼─────────┼─────────┤
│ CollisionTest         │ SUCCESS │ 0 times │
├───────────────────────┼─────────┼─────────┤
│ RunningRedLightTest   │ SUCCESS │ 0 times │
├───────────────────────┼─────────┼─────────┤
│ RunningStopTest       │ SUCCESS │ 0 times │
├───────────────────────┼─────────┼─────────┤
│ InRouteTest           │ SUCCESS │         │
├───────────────────────┼─────────┼─────────┤
│ AgentBlockedTest      │ FAILURE │         │
├───────────────────────┼─────────┼─────────┤
│ Timeout               │ SUCCESS │         │
╘═══════════════════════╧═════════╧═════════╛

> Registering the route statistics
ERROR: failed to destroy actor 414 : unable to destroy actor: not found 
ERROR: failed to destroy actor 436 : unable to destroy actor: not found 
ERROR: failed to destroy actor 551 : unable to destroy actor: not found 
ERROR: failed to destroy actor 684 : unable to destroy actor: not found
```

LeBronLiHD, 2022-12-23:
any help will be greatly appreciated!!!

LeBronLiHD, 2022-12-23:
I hope this is not because I only kept one model under `model_ckpt/transfuser/`

```python
$ ls -al model_ckpt/transfuser/
total 657600
drwxr-xr-x 2 haodong haodong      4096 Dec 23 23:30 .
drwxrwxr-x 6 haodong haodong      4096 Dec 23 23:30 ..
-rw-r--r-- 1 haodong haodong       542 Jan  8  2022 args.txt
-rw-r--r-- 1 haodong haodong 673362276 Jan  8  2022 model_seed3_37.pth
```

Kait0, 2022-12-23:
One model is fine. (Though your loading message above shows a different one: `model_seed1_39.pth` than your ls command: `model_seed3_37.pth`

I would recommend you turn on the debug outputs to see if the inputs are correct, and what the model predictions do.
You need to set the environment variable [SAFE_PATH](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/submission_agent.py#L23).
and turn [debug ](https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/config.py#L35)to true:

This will generate some debug images in the SAVE_PATH.
You can post some here.

In general it is possible that the model is failing because of the distribution shift coming from `-quality-level=Low` that you mentioned earlier.
That setting is quite drastic for example the sky is not rendered at all anymore.

Somehow your original problem seemed to be that CARLA/rendering is interfering with cuda.
I never had that problem so I don't know how to solve it properly.
CARLA has two rendering options -opengl and -vulkan. You can try both.
Also you can try turning off the screen rendering with:

> SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh -world-port=2000 -opengl

LeBronLiHD, 2022-12-24:
[image]

dear developer, the problem was solved by using `-vulkan` rendering option

I am super grateful for your kind help!

closed 2022-12-24 by LeBronLiHD