autonomousvision/transfuser #137: Is the CARLA Leaderboard still available on Alpha Drive?

KP-Zhang, 2022-12-28:
Thank you for your code. It is very helpful.
We have already trained our model on our local computer. We are going to evaluate our model in the CARLA Leaderboard via Alpha Drive. To submit our Docker image, we created a team last week. However, our team still have not been approved. Is the CARLA Leaderboard still available?
[image]
Thank you for your time.

Kait0, 2022-12-29:
As far as I am aware the leaderboard is still available.
The tutorial for the submission can be found [here](https://leaderboard.carla.org/submit_v1/).
For problems regarding the leaderboard you should contact the maintainers.
Guillermo López, Joel Moriana or German Ros are the people you need to talk to.
You can do that via their [e-mail](https://leaderboard.carla.org/help/) or the [CARLA discord](https://discord.com/invite/8kqACuC)
In Europe we currently have the christmas holidays and people are usually on vacation.
I presume that this is the reason why your approval takes so long.

closed 2022-12-29 by Kait0

KP-Zhang, 2022-12-30:
Thanks a lot for your information.