autonomousvision/transfuser #140: Shape mismatch in GPY

BXYMartin, 2023-01-11:


Kait0, 2023-01-12:
That is a bug that unintentionally mixes up the channel and pixel dimensions.
Since the network can in principle learn to deal with it, and I'm not sure if it causes any problems, we left it in.
Changing it would break the pre-trained models.

As you mentioned, the problem is that at the point where the view gets applied, the pixel dimensions are still in front of the channel dimension due to the transformation earlier:
`image_tensor = image_tensor.view(bz, self.seq_len, -1, img_h, img_w).permute(0,1,3,4,2).contiguous().view(bz, -1, self.n_embd)`

I think the correct implementation should be:

```
image_tensor_out = x[:, :self.seq_len * self.config.img_vert_anchors * self.config.img_horz_anchors, :].view(
        bz * self.seq_len, img_h, img_w, -1).permute(0, 3, 1, 2).contiguous()
lidar_tensor_out = x[:, self.seq_len * self.config.img_vert_anchors * self.config.img_horz_anchors:, :].view(
          bz * self.seq_len, lidar_h, lidar_w, -1).permute(0, 3, 1, 2).contiguous()
```

BXYMartin, 2023-01-12:
Yes, that explained my doubt, thank you so much!

closed 2023-01-12 by BXYMartin