autonomousvision/transfuser #141: How many epochs should I train to get a well performed TransFuser Model.

Hongkuan-Zhou, 2023-01-16:
Dear Authors,
I am training the transfuser model with the '02_05_withheld' setting cause I would like to test the model in the town05 long benchmark. I trained for 30 epochs. However, the testing result shows that the agent can barely finish the task. It always runs out of the street.

Best wishes
Hongkuan Zhou

Kait0, 2023-01-16:
31 epochs is what we usually use, because the learning rate gets reduced at epoch 30.

Hongkuan-Zhou, 2023-01-17:
The model I trained with the '02_05_withheld' setup for 30 epochs still can't complete the task (the RC is only 5%). The agent always hits the statics when turning. Could I ask if you could provide the model trained with '02_05_withheld'?
Best wishes
Hongkuan zhou

closed 2023-01-17 by Hongkuan-Zhou