autonomousvision/transfuser #142: Problem by evaluation

JianLiMech, 2023-01-24:
Thank you for your great work. I have been trying to evaluate my model after training. 
But when I try to evaluate it, this happens:
[image]

I have set `RESUME=0` in local_evaluation.sh, but it doesn't work. Thank you in advance!

JianLiMech, 2023-01-25:
> 

Thanks for your quick reply.
The local_evaluation.sh setup looks like this:
```
export CARLA_ROOT=${1:-~/Documents/transfuser/carla}
export WORK_DIR=${2:-~/Documents/transfuser}

export CARLA_SERVER=${CARLA_ROOT}/CarlaUE4.sh
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg
export SCENARIO_RUNNER_ROOT=${WORK_DIR}/scenario_runner
export LEADERBOARD_ROOT=${WORK_DIR}/leaderboard
export PYTHONPATH="${CARLA_ROOT}/PythonAPI/carla/":"${SCENARIO_RUNNER_ROOT}":"${LEADERBOARD_ROOT}":${PYTHONPATH}

export SCENARIOS=${WORK_DIR}/leaderboard/data/longest6/eval_scenarios.json
export ROUTES=${WORK_DIR}/leaderboard/data/longest6/longest6.xml
export REPETITIONS=1
export CHALLENGE_TRACK_CODENAME=SENSORS
export CHECKPOINT_ENDPOINT=${WORK_DIR}/results/transfuser_longest6.json
export TEAM_AGENT=${WORK_DIR}/team_code_transfuser/submission_agent.py
export TEAM_CONFIG=${WORK_DIR}/model_ckpt/transfuser
export DEBUG_CHALLENGE=0
export RESUME=0
export DATAGEN=0

python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator_local.py \
--scenarios=${SCENARIOS}  \
--routes=${ROUTES} \
--repetitions=${REPETITIONS} \
--track=${CHALLENGE_TRACK_CODENAME} \
--checkpoint=${CHECKPOINT_ENDPOINT} \
--agent=${TEAM_AGENT} \
--agent-config=${TEAM_CONFIG} \
--debug=${DEBUG_CHALLENGE} \
--resume=${RESUME}
```
I still don't get why these is no route to drive.

Kait0, 2023-01-25:
Looks fine to me. You can try deleting 
transfuser_longest6.json
Otherwise you will need to debug what is going on in the pyhton file.

JianLiMech, 2023-01-25:
> Looks fine to me. You can try deleting transfuser_longest6.json Otherwise you will need to debug what is going on in the pyhton file.

Thank you, I solved this by deleting transfuser_longest6.json, I will close this issue.

closed 2023-01-25 by JianLiMech