autonomousvision/transfuser #143: Questions regarding to the size of dataset

JianLiMech, 2023-02-09:
Hello,

thanks for your great work! 

I am trying to train a model according to your dataset. But our computational resource is not powerful enough to handle 200GB dataset. We have 4 2080Tis and it will cost over 2 weeks to train a model with your dataset. That would be too long. So I would like to ask you, if you know how much data would be adequate to train a relative good model (reach approximately 80% of your driving score)? It would be great help if you could answer this question. Thank you in advance.

JianLiMech, 2023-02-09:
I tried to train a 20GB dataset. The result was kind of bad. The route completion was 29% and the driving score was 3.

Kait0, 2023-02-09:
> 4 2080Tis and it will cost over 2 weeks to train

This seems excessive to me.
We trained our models with 8x 2080Ti in 1 day.
Maybe you should profile your system to see if there is a bottleneck somewhere slowing the whole training down.

I don't have any insight in how much data the data can be reduced. 
We never ran an experiment on that.

JianLiMech, 2023-02-12:
Thanks for replying.
I have some other questions:
1, What command should I use in terminal, if I want to load file from the last training?
I tried 
`CUDA_VISIBLE_DEVICES=0,3 OMP_NUM_THREADS=32 OPENBLAS_NUM_THREADS=1 torchrun --nnodes=1 --nproc_per_node=2 --max_restarts=0 --rdzv_id=1234576890 --rdzv_backend=c10d train.py --logdir /mrtstorage/users/jianli/TF/log_dir --load_file /mrtstorage/users/jianli/TF/log_dir/model_11.pth --start_epoch 11 --root_dir /mrtstorage/users/jianli/TF/data/ --parallel_training 1`
But the Error said No such file` /mrtstorage/users/jianli/TF/log_dir/model_11.pth. ` Did I use to wrong command to load file or should I just delete `--load_file /mrtstorage/users/jianli/TF/log_dir/model_11.pth` and the file will automatically be loaded?

2, How can I get the event file when using tensorboard? I tried to use tensorboard but no  event file was found.

[image]
[image]

Kait0, 2023-02-13:
Your path is simply wrong. It should be:
--load_file /mrtstorage/users/jianli/TF/log_dir/transfuser/model_11.pth
As you can also see in your second screenshot.
The training script creates a folder in the log dir with the training id name.

You start tensorboard with:
tensorboard --logdir "/mrtstorage/users/jianli/TF/log_dir"

JianLiMech, 2023-02-14:
Thank you, everything works now! I will close this issue.