autonomousvision/transfuser #146: Dying RELUs in transformer layers

pimpke, 2023-02-27:
Hi, I'm having training issues as many RELU neuron units in the transformer layers are "dead" (outputting 0 for all the training examples). This happens only for RELUs in those layers, image encoder and lidar encoder RELUs are looking good (near 0% of dead neurons). Here are the RELUs from transformer 4:

| RELU layer name                                 | % of dead neurons |
| -------------------------------------------------- | ----------- |
| encoder.transformer4.blocks.0.mlp.0   | 0.           |
| encoder.transformer4.blocks.0.mlp.1   | 6.83       |
| encoder.transformer4.blocks.0.mlp.2   | 45.86     |
| encoder.transformer4.blocks.0.mlp.3   | 59.77     |
| encoder.transformer4.blocks.0.mlp.4   | 69.19     |
| encoder.transformer4.blocks.0.mlp.5   | 72.05     |
| encoder.transformer4.blocks.0.mlp.6   | 72.90     |
| encoder.transformer4.blocks.0.mlp.7   | 73.76     |

Transformers 1-3 are following a similar pattern (blocks.0 starts with 0% dead neurons and the following blocks have an increased number of dead neurons - up to ~70% for the last block).

Have you maybe had a similar training issue, and if so, how did you resolve it? :smile_cat: 

Thanks!

kashyap7x, 2023-03-01:
Could you please repeat this analysis for the 3 publicly released TransFuser checkpoints? https://github.com/autonomousvision/transfuser#pretrained-agents

It isn't clear to me why sparsity in the transformers is necessarily bad. If you look at Fig. 2 in https://arxiv.org/pdf/2205.15997.pdf, notice how the transformers act as skip connections to refine the features of the Image and BEV branches, so it is not necessary to have dense non-zero outputs for tokens where the CNN features are already useful.

If the pre-trained models provided by us show a similar sparsity distribution, there is probably no issue with your training.

pimpke, 2023-03-02:
Hi, I ran the analysis for those checkpoints, and they're looking much better - less than 10% of dead ReLUs for almost all transformers' ReLU layers.

Kait0, 2023-03-03:
the released were trained with an earlier version of the training code.
I uploaded one trained with the current code here: https://drive.google.com/drive/folders/1oPiC88CcUcfoOzBon4n8gYcAD8QT_23r?usp=sharing
If you also don't see the problem there, then there is probably something wrong with your training setup.

closed 2023-03-14 by Kait0

matthijsvk, 2023-06-29:
> Hi, I'm having training issues as many RELU neuron units in the transformer layers are "dead" (outputting 0 for all the training examples). This happens only for RELUs in those layers, image encoder [...]

how did you detect the dead neurons?