autonomousvision/transfuser #150: ALSA lib error

EcustBoy, 2023-03-29:
Hi~ I try to run transfuser-2022 code on my server without screen interface, I use carla 0.9.12 server in docker and use carla0.9.12 client in conda env. I run as transfuser official instructions, and the carla server and client can launch and connect normally, and scenario or routes can be loaded normally too, but then the following error occurs:
[image]
 I'm guessing if this is related to pygame render settings, could you provide some suggestions?

Kait0, 2023-03-29:
ALSA is a sound library. The errors probably come because you don't have a sound card or it is not connected to your docker.
You can start CARLA with the  
`-nosound`
option than these errors should disappear.
We don't use any sound info in Transfuser

EcustBoy, 2023-03-30:
> ALSA is a sound library. The errors probably come because you don't have a sound card or it is not connected to your docker. You can start CARLA with the `-nosound` option than these errors should disappear. We don't use any sound info in Transfuser

Hi author~ thanks for your advice, it indeed avoid the docker interrupt error after add '--nosound',  but another error occured:
I run datagen.sh for data collecting, I haven't changed any params in this script except for the specific path param，and the first scenario can finish normally, 
[image]

but when it reload the second scenario, the following error occurs:
[image]
and then the simulation stop:
[image]
I try different scenario in different town, but this error always existe.  I guess if this bug has something to do with reloading town during simulation?  Or due to my personal machine configuration?  So I wanna know:
(1) Have you ever encountered this type of error? Is there any advice provided to me？
(2) Where the data(e.g.  pictures, map, odom information, etc) collected during simulation is stored, I only find the json files in /results folder
I will appreciate your response~

Kait0, 2023-03-31:
This error shouldn't happen consistently so it is some problem with your setup.
CARLA sends sensor data over the streaming port, maybe you should set it explicitly when starting the serve and make sure it's free.:

>  CarlaUE4.sh -carla-rpc-port=${FREE_WORLD_PORT} -nosound -carla-streaming-port=${FREE_STREAMING_PORT} -opengl

EcustBoy, 2023-03-31:
> This error shouldn't happen consistently so it is some problem with your setup. CARLA sends sensor data over the streaming port, maybe you should set it explicitly when starting the serve and make [...]

Hi~ my carla server (in docker) launch command is:
docker run -p 2000:2000 -p 2001:2001 --privileged --gpus '"device=0"' --net=host carlasim/carla:0.9.12 /bin/bash ./CarlaUE4.sh -RenderOffScreen --carla-rpc-port=2000 --carla-streaming-port=2001 --trafficManagerPort=8000
it seems normal right? and L'm sure these ports are all free before docker launch

Kait0, 2023-03-31:
The CARLA options for the ports you use with double --, I think single - is correct for carla-rpc-port and carla-rpc-port.
traffic manager port is not an option for CARLA as far as I am aware. That port gets communicated via the client.
(CARLA ignores options it doesn't know).

Not sure if it is relevant but you might want to also expose port 8000.

I vaguely remember that I had problems with using consqutive ports once. I would suggest you try something like 2050 for the streaming port.

you are using carla 0.9.12. I am not sure if that is still compatible with this codebase which is build for 0.9.10.1.
I think I tested 0.9.13 once and it was not compatible anymore.