autonomousvision/transfuser #156: Branch 2021 Dataset Download Error

jimmyfyx, 2023-04-18:
Dear Authors,

When we were trying to download the dataset from the 2021 branch, we met an unexpected error as below:
[image]
We have successfully downloaded the data for the first four town, but fail to download the data starting from Town05. As you see in the screenshot, after encountering the error when running `./download_data.sh`, we directly copy the command in the .sh file and run it in the terminal, but the error still exists.

I wonder is there any solution for this issue? Thanks so much!

kashyap7x, 2023-04-18:
Could you try again? https://s3.eu-central-1.amazonaws.com/avg-projects/transfuser/data/14_weathers_data/Town05.zip is currently working for me. It might have been a temporary AWS outage

jimmyfyx, 2023-04-18:
Thanks for the reply! I think it works now

closed 2023-04-23 by Kait0