autonomousvision/transfuser #157: agent get blocked with zombie vehicles

raozhongyu, 2023-04-18:
I test my own code on longest6  while I meet a lot of zomblie vehicles get blocked.  Could you give me some suggestions?

Kait0, 2023-04-18:
You would need to provide more details on your problem. What do you mean by zombie vehicle?

raozhongyu, 2023-04-18:
The zombie vehicle is the other vehicles which set the autonomous driving. I notice the vehicle may get blocked somtimes,  such as no move or block at the  crossroad.  Is that normal ?

Kait0, 2023-04-18:
If you mean the other traffic can get stuck in intersections this is normal yes.
The algorithm that drives them is not perfect.
Other vehicles will de-spawn randomly if they are stuck for more then 90s which usually resolves this problem.

raozhongyu, 2023-04-18:
Thanks a lot

closed 2023-04-18 by raozhongyu