autonomousvision/transfuser #161: How can I view the BEV  and point cloud

QingZhuanya, 2023-05-05:
good work!
But how can I view the top-down view and point cloud of vehicles as well as its trajectories in the evaluation dataset?
like [this:](https://www.bilibili.com/video/BV19W4y1q7nm/?spm_id_from=333.788.recommend_more_video.8&vd_source=9011820b8f246d5c448e55fdf5f996a6)

Kait0, 2023-05-05:
https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/model.py#L935
https://github.com/autonomousvision/transfuser/blob/a7d4db684c160095dec03851aff5ce92e36b2387/team_code_transfuser/model.py#L719
Set the debug flag and a SAVE_PATH

QingZhuanya, 2023-05-06:
I have run local_evaluation.sh，but Carla still only has a top-down view.
like this:
[image]

closed 2023-05-08 by QingZhuanya