autonomousvision/transfuser #162: Car stops constantly

mmahdavian, 2023-05-09:
Hello. I have trained a model using your codes and dataset and when I want to test one of the model.pth files, the car doesn't move in Carla or goes to a wrong point. I also checked my codes with your pre-trained model. When it sees all three model files in the transfuser model, it moves well. But if it only sees one of them it doesn't work. 
Which ones of my trained model files should the submission agent see to move correctly as there are several model and optimization files.

Thank You

Kait0, 2023-05-10:
Epoch 31 (you only need the model not the optimizer) is usually a good file to use and you need one args.txt file in the config folder to load the configuration.
All of the three model files we released should work (you always need an args.txt, + n>0 model files), if not there is some problem with your setup.
If you trained without parallel than maybe https://github.com/autonomousvision/transfuser/issues/110 is related to your problem?

mmahdavian, 2023-05-10:
Thank you. It fixed the problem. I think this a important change and needs to be mentioned in the first page.

closed 2023-05-10 by mmahdavian