autonomousvision/transfuser #164: Questions about training loss

SmashChen, 2023-05-20:
Dear Authors,
  Thank you for your great work! In PAMI22,  could you offer me the approximate loss in training and validation for waypoints and other auxiliary tasks?  Thank you very much!

Kait0, 2023-05-22:
We didn't use validation losses for the experiments in the paper as we trained on all the data.
Training losses were somethink like:
Loss_bev: 0.13
loss_center_heatmap: 0.03
loss_depth: 0.199
loss_offset: 0.02
loss_seg: 0.05
loss_total: 0.99
loss_wh: 0.01
loss_wp: 0.04
loss_yaw_class: 0.02
loss_yaw_res: 0.0005

Open loop losses are not really indicative of closed loop performance, I typically don't pay much attention to them.
But you can see whether there is some optimization problem, losses should always go down.
Should look something like this:

[image]

SmashChen, 2023-05-23:
Thank you very much for your reply!

closed 2023-05-24 by Kait0