autonomousvision/transfuser #168: question about carla simulation

EcustBoy, 2023-05-27:
Hi author, I tried to evaluation my own model in carla simulator with town05_short routes, I follow the instructions to use transfuser-2021 branch code and launch simulation, but it seems the route waypoints aren't spawned correctly, I plot town05_short waypoints in map:
[image]
you can see some start or goal waypoint is out of drivable area, such that the ego car may block during simulation because it thinks it reach the final goal but actually not. I'm sure I directly copy the raw route xml files as you suggest from transfuser-2021 (And when I visualize town05_long route xml files, it seems normal.)

So I wanna ask have you met such strange problem? Is it possible due to carla version? (I notice your project use carla-0.9.10 while I use carla-0.9.12).

This bug really makes me very confused  , I'll thanks much for your reply~

Kait0, 2023-05-27:
I would first make sure this isn't a visualization problem.
I think the leaderboard client projects waypoints in the route file onto the nearest lane.
You can visualize the route in CARLA by seeting --debug 1 in the leaderboard_evaluator.py arguments.

EcustBoy, 2023-05-28:
> I would first make sure this isn't a visualization problem. I think the leaderboard client projects waypoints in the route file onto the nearest lane. You can visualize the route in CARLA by seeting --debug 1 in the leaderboard_evaluator.py arguments.

Thanks for your reply~~ I'm quite sure this isn't a visualization code bug, because I also check each start and goal waypoints position from the xml file on global map, some waypoints are indeed not in a valid drivable range.

Although client can projects waypoints in the route file onto the nearest lane and client will treat the projected waypoint as a goal, but as I think obviously this is not reasonable.  I'm not sure if there's something wrong with the official town05_short file :-) At least I didn't find any bug in my own code.

So I ‘d like to know if you encountered such an exception during the simulation? Is your simulation car normally completing the entire route? (I mean running exactly from start coordinates to goal coordinates descripted by the route xml files)

I'll thanks much for your answer

Kait0, 2023-05-30:
The route xml files are autogenerated and don't necessarily start or end on the road (as said CARLA will project the points to the road). 
The agent gets the projected points as input not the original xml data, so this isn't really a problem.
I don't think there is a bug in your code, these route files are just not so clean.

closed 2023-05-30 by Kait0