autonomousvision/transfuser #169: About train

cstlove, 2023-05-27:
Thank you so much for your outstanding work!
I used the model weights file you provided with good results.
I tried to use the code and training set you provided for training, but there were some problems. After nearly 40 rounds of training, the model did not converge (due to equipment limitations, I adjusted the batch size to 6). The number of training is still Is it not enough or because of the batch size?

[image]

Kait0, 2023-05-27:
I wouldn't train for longer, training loss should not go up during training. Batch size 6 is quite small.
Perhaps you can try to reduce the learning rate. It's a trick in the community to do that when scaling down batch size, because you are gonna make more gradient steps due to the smaller batch size.
Also make sure to adjust your inference file if you train with a single GPU (https://github.com/autonomousvision/transfuser/issues/110)

ChaselWan, 2023-05-30:
hello!Nice to see your train works well,I have a question about root_dir in train.py.
The root_dir of dataset should be set in "/s1" or "/s1/dirt_dataset" or something else?
Looking forward to your reply!Thanks very much!

Kait0, 2023-05-30:
The very root folder containing all other folders.
I guess that would be s1 in your case?

The code will crash if you use the wrong folder.

ChaselWan, 2023-06-01:
> The very root folder containing all other folders. I guess that would be s1 in your case?
> 
> The code will crash if you use the wrong folder.

I have tried this folder,but error says:
[image]

I check codes in `config.py`,line208 `if(setting==all)  # All towns used for training no validation data`
Maybe the dataset root folder should contain all towns I guess? The following is what S1 contains:
`S1`
`-dataset_root_23_11`
`--Routes_Scenario1_Town01_curved_Seed0`
`---Scenario1_Town01_curved_route0_11_23_20_00_55`
`----depth、lidar and so on`
Should I zip all towns in one folder and make it my root folder?
Looking forward your reply!

ChaselWan, 2023-06-05:
> 

Thanks！I would try it again.

closed 2023-06-12 by Kait0