autonomousvision/transfuser #170: question about collision avoidance in carla

EcustBoy, 2023-05-27:
Hi author, I design my own end-to-end model (only takes 6v camera rgb as input without lidar)，first training in open-loop manner, and then launch close-loop simulation in carla as your instruction. 

But I found my model is **_not well learned in collision avoidance_**, such as :  not always being able to reasonably slow down and stop when there is a vehicle ahead, sometimes there will be a delay reaction , but such delay can lead to collision. Or even not being to respond to other cars. 

I collected my own training dataset (about **_75k_**,  cover multiple scenarios on each public town map except town05) from carla-0.9.12 by autopilot agent,  using the same code as your transfuser project. And  I'm sure my dataset format is correct by visualization, without any low-level errors (e.g.  ego car gt trajectory error)  

I wanna ask is it possible simply due to **_my dataset size is not enough such that my model can't learn the basic behavior_**？
Have you met such problem during research,  have you investigated **the impact of different scale datasets on model's close-loop ability**? 
I was really puzzled by this anomaly in my experiment,  I'll thanks much if you can provide some insight or suggestions

Kait0, 2023-05-27:
Perhaps also visualize some of these errors and see if the cars the vehicle crashed into were properly visible in the cameras.
Not well learned is a bit vague. TF also has collision infractions. Is it drastically different from the released model?

I think the scenario runner becomes incompatible with newer carla versions, I'm not sure if 9.12 or 9.13 was the breaking release.

Yes dataset scale is certainly important.
There is a controlled study on the importance of dataset size in our groups latest work [PlanT](https://arxiv.org/abs/2210.14222) Table 2b). 
It investigates what happens if you scale up compared to this repos dataset. 
I think you can extrapolate these results into the other direction as well.
The plot is about priviledged planning but the results are similar for sensori-motor agents.

EcustBoy, 2023-05-30:
> Perhaps also visualize some of these errors and see if the cars the vehicle crashed into were properly visible in the cameras. Not well learned is a bit vague. TF also has collision infractions. Is [...]

Thanks for your reply~  I wanna ask something else.  First,  I try to run your transfuser-2022 code with my carla-0.9.12 server,  but I found in some area, traffic flow can be blocked by themself.  Then, I follow your advice, choose scenario-runner-0.9.12 to replace original scenario_runnner-0.9.10 in your code, but sometimes the traffic block bug is still there. 

So I wanna ask does the leaderboard package version also have to match with Carla server version? Maybe I should choose a newer commit version of leaderboard which is compatible to carla-0.9.12?

Kait0, 2023-05-30:
That the traffic flow can be blocked happens also in carla-0.9.10
CARLA despawns cars that are stuck for over 90 - 180 seconds. (in some CARLA version after 9.10 this was increased from 90 to 180 depending on the situation, don't remember which one).

Second thing that comes to mind is that the leaderboard client changed the block infraction to happen after 180 at the end of 2020. We are using that version here (some older version gave block infractions after 90 seconds).

So in our setting we get blocked after 180 seconds standing still, but other cars are despawned after 90 seconds, so traffic congestion is usually not a problem (other cars will despawn in time).

EcustBoy, 2023-06-01:
> That the traffic flow can be blocked happens also in carla-0.9.10 CARLA despawns cars that are stuck for over 90 - 180 seconds. (in some CARLA version after 9.10 this was increased from 90 to 180 [...]

Many thanks for your reply~~ By the way, I plan to do some research about the interaction between ego planning and other cars' prediction, but I notice the current traffic simulation behavior is relatively simple and conservative.  

So I wanna ask have you tried to adjust the traffic agents' or ego car's behavior style during data collection or evaluation? 
e.g. **enable other agents to lane change,  or set a probability to decide whether ego/other car is aggresive or cautious by adjusting autopilot model or traffic manager parameter**?
(actually I'm not quite familiar about how to adjust autopilot param)

 I think maybe such method can improve model's robustness and dataset diversity?

Kait0, 2023-06-03:
No we have not adjusted the traffic manager.
As for the benchmarks we look at the other cars are behaving the same during evaluation as during data collection.
Not sure if there is some advantage of making the traffic during data collection different than during evaluation.

closed 2023-06-12 by Kait0