autonomousvision/transfuser #172: Questions about train set and training

JinRanYAO, 2023-06-13:
Hello, thank you for your excellent work. I try to train the transfuer model using Town 03 and 04, with batch size 16 and 4 GPUs. But I found that it seems underfitting, and the model has bad behaviour in CARLA Longest 6.
The Training curve for loss total and waypoints:
[image]
So I have some questions:
1. What do you think is the reason for this? Is the train set too small, or is it caused by changing the batch size to 16?
2. What do you think I should improve next? Is it adding more epochs, or adding more training data?
3. If I want to get the ideal behaviour in CARLA, what is the minimum train set setting?
4. Is the pre-trained model provided in this repository trained with the whole open-sourced dataset? Is there more data?
I am sorry to bother you, and looking forward to your reply!

Kait0, 2023-06-13:
Traffic light differ between Town 01/ 02 and the rest of the towns.
If you only train on Town 03/04 you basically fail at red lights by design, you should at least use either Town 01 or 02.

What does bad behavior mean?

1. The lowest batch sizes I have used were ~40 and that did not seem to cause problems. You training set is definitely too small you should train on all towns. Why are you not using a higher batch size if you have 4 GPUs? Or do you mean batch size of 16 per GPU?
2. You don't need more epochs the 40 you have here is enough. I would expect more data to help unless there is another problem here.

3. If you want to reproduce the results in the paper you need to train on all of the dataset.
4. Yes the pre-trained model is trained with the entire dataset. There is not more data.

Besides open loop losses are not really indicative on closed-loop performance.

JinRanYAO, 2023-06-13:
Hello, thank you for your reply!

> Traffic light differ between Town 01/ 02 and the rest of the towns. If you only train on Town 03/04 you basically fail at red lights by design, you should at least use either Town 01 or 02.
> 
> What does bad behavior mean?

It doesn't just fail the red light test, it also fails to maintain accurate lanes and does not drive in the middle of the lane, does not turn into the lane accurately when turning at an intersection, causing collisions with other vehicles.

> 1. The lowest batch sizes I have used were ~40 and that did not seem to cause problems. You training set is definitely too small you should train on all towns. Why are you not using a higher batch size if you have 4 GPUs? Or do you mean batch size of 16 per GPU?

I'm sorry I didn't explain this clearly. I have a batch size of 16 for each GPU.

> 2. You don't need more epochs the 40 you have here is enough. I would expect more data to help unless there is another problem here.
> 3. If you want to reproduce the results in the paper you need to [...]

Thank you for your advice, and I will try to use more data for training.

> Besides open loop losses are not really indicative on closed-loop performance.

Do you think the abnormal behavior in CARLA I just mentioned are also caused by insufficient training data? Or could there be other errors?

Kait0, 2023-06-14:
> Do you think the abnormal behavior in CARLA I just mentioned are also caused by insufficient training data? Or could there be other errors?

Hard to tell but low amounts of data could cause these issues, yes.