autonomousvision/transfuser #174: Questions about transfuser_longest6.json file parsing

carryman97, 2023-06-27:
Hello, your work is excellent and I am very interested in it. However, I recently encountered a problem. I generated the transfuser_longest6.json file by performing performance tests on the longest 6 routes, but when I run result_parser.py repeatedly, there will be different driving scores, and I am very puzzled. I don't know which driving score is the real driving score. I want to ask you, can I read from transfuser_longest6.json to get real driving score? Does "score_composed" in transfuser_longest6.json represent driving score? If yes, I would like to ask if the top "score_composed" in transfuser_longest6.json represents the average driving score of 36 routes?
 Looking forward to your answer, your answer will help me solve the current doubts. Thanks

Kait0, 2023-06-27:
running the result parser should give you the same result csv file every time.
Can you send us the transfuser_longest6.json and the commands you used to run the result parser, so that we can reproduce the problem?

carryman97, 2023-06-27:
When I run result_parser.py multiple times under the same project file, there will be two driving scores, 38 and 35 respectively. When I copy transfuser_longest6.json to another project file (this project is also "transfuser", I back it up I created multiple "transfuser" projects to facilitate more experiments), the driving score is 47.93, I don't know which one is real, can you directly see the real driving score through the transfuser_longest6.json file, thank you!W

crazyman
***@***.***

Kait0, 2023-06-27:
The result json also puts an average DS of the routes in this file at the end of the file.
However only the DS, RC, IS are correct in this file, the other metrics are calculated wrongly (that is why you need the result parser, it will recalculate them with the correct math).
The result parser should produce the same DS as written in there.
As far as I can see the result parser is deterministic, so I don't see how you can get different results.
Can you email me an example .json file where this happens (bernhard.jaeger@uni-tuebingen.de)?
I would like to reproduce the error to see what is going on.

Note that the result parser will average all .json files inside the folder you give him, so you should make sure that there is only 1 file (that contains all routes) or there are 36 (* repetiotion) .json files each containing 1 route.

closed 2023-06-28 by Kait0