autonomousvision/transfuser #175: question about validation dataset and evaluation using pretrained models

JinRanYAO, 2023-07-01:
Hello, I am sorry to bother you again. I have two questions now:
1.If I want to reproduce the same training procedure as yours, which setting should I use, 'all' or '02_05_withheld'? Should I keep towns 02 and 05 for validation, and then choose the model with the lowest valid loss in 31 33 35 37 39 41 epochs?
2.I evaluate the 3 public pretrained transfuser model, model_seed1_39.pth, model_seed2_39.pth, model_seed3_37.pth, but I can not get the same result as table1 in TransFuser paper:
`"values": [
        "43.978",
        "82.842",
        "0.551",
        "0.004",
        "0.641",
        "0.041",
        "0.060",
        "0.094",
        "0.041",
        "0.000",
        "0.103",
        "0.165"
    ]`
`"values": [
        "44.727",
        "83.393",
        "0.534",
        "0.010",
        "0.684",
        "0.027",
        "0.027",
        "0.154",
        "0.021",
        "0.000",
        "0.025",
        "0.211"
    ]`
`"values": [
        "35.669",
        "71.854",
        "0.544",
        "0.004",
        "1.081",
        "0.036",
        "0.053",
        "0.092",
        "0.243",
        "0.000",
        "0.038",
        "0.573"
    ]`
but in table1:
[image]
Is it normal or something wrong?

Kait0, 2023-07-01:
1. all experiments in the paper were done with the 'all' setting. 
 '02_05_withheld' is an extra feature if you want to use other benchmarks like LAV.
Validation loss is pretty meaningless. We either used epoch 31 which is always close to the best, or evaluated all 6 epochs closed loop on the NEAT routes and picked the best.

2. The result in Table 1 is an ensemble. You need to evaluate the 3 models together not individually (just put the 3 model files into the same folder the code will ensemble them automatically.
3. You need to use the result parser to process the results to get the correct auxiliary metrics.

Lastly you won't get exactly the same result. As you can see the std of +- 6 is quite large

JinRanYAO, 2023-07-02:
Thanks for your reply!

> We either used epoch 31 which is always close to the best

But why are 39.pth, 39.pth, 37.pth in the public pretrained transfuser model?

> ensemble

Does ensemble always improve performance, or reduce performance sometimes? Can ensemble be performed on different epochs in the same training process, such as among seed1_31.pth, seed1_33.pth and seed1_35.pth?

> std of +- 6

Approximately how many evaluations will have a std of 6?

Kait0, 2023-07-02:
> But why are 39.pth, 39.pth, 37.pth in the public pretrained transfuser model?

For table 1 we evaluated all 6 epochs on NEAT. For the ablations we used epoch 31.

> Does ensemble always improve performance, or reduce performance sometimes? Can ensemble be performed on different epochs in the same training process, such as among seed1_31.pth, seed1_33.pth and seed1_35.pth?

I don't think it will reduce performance (significantly) but sometimes it makes no difference.
We have been using epochs from different training runs. I don't know if epochs from the same training run will achieve the same effect (probably not as the models are not different enough).

> Approximately how many evaluations will have a std of 6?
The std is evaluation variance. We used 3 repetitions of the longest6 routes for all experiments.

JinRanYAO, 2023-07-02:
Oh ok, I see. Thank you very much for your reply!

closed 2023-07-02 by JinRanYAO