autonomousvision/transfuser #183: normalize the gps

a1wj1, 2023-07-26:
Hi -- thanks for all your work in releasing this code and that for the baselines. I noticed that in the RoutePlanner you have a mean and scale that is used to normalize the gps from the sensor.

It seems to have 2 values for 2 versions of carla. I'm trying to use this with a different version of carla (9.11) so I'm trying to find the equivalent.
```
        # self.mean = np.array([49.0, 8.0]) # for carla 9.9
        # self.scale = np.array([111324.60662786, 73032.1570362]) # for carla 9.9
        self.mean = np.array([0.0, 0.0]) # for carla 9.10
        self.scale = np.array([111324.60662786, 111319.490945]) # for carla 9.10
```
Thanks!

I have understood the calculation method of mean and use the following function to calculate：

```
def _get_latlon_ref(world):
    """
    Convert from waypoints world coordinates to CARLA GPS coordinates
    :return: tuple with lat and lon coordinates
    """
    xodr = world.get_map().to_opendrive()
    tree = ET.ElementTree(ET.fromstring(xodr.encode('utf-8')))

    # default reference
    lat_ref = 42.0
    lon_ref = 2.0

    for opendrive in tree.iter("OpenDRIVE"):
        for header in opendrive.iter("header"):
            for georef in header.iter("geoReference"):
                if georef.text:
                    str_list = georef.text.split(' ')
                    for item in str_list:
                        if '+lat_0' in item:
                            lat_ref = float(item.split('=')[1])
                        if '+lon_0' in item:
                            lon_ref = float(item.split('=')[1])
    return lat_ref, lon_ref
```

But I'm not sure about the calculation method for scale yet. Can you help explain it？

Kait0, 2023-07-26:
These numbers are still from the carla leaderboard starter kit.
I don't know how they were computed. I assume someone collected a dataset of ground truth, gps pairs and ran linear regression on it.
I tested other mathematical conversion methods, but the results were always biased and much less accurate than this method which has less than 1 cm conversion error.

You should test first whether these values still work in CARLA 0.9.11 before trying to make new once.

a1wj1, 2023-07-27:
OK，Thank you very much！

closed 2023-07-28 by Kait0