autonomousvision/transfuser #193: About the dataset

Lesliewsq, 2023-10-10:
I'm in China，how should I download the dataset？
```
wget https://s3.eu-central-1.amazonaws.com/avg-projects/transfuser/data/14_weathers_data/Town05.zip
--2023-10-10 10:18:56--  https://s3.eu-central-1.amazonaws.com/avg-projects/transfuser/data/14_weathers_data/Town05.zip
Resolving s3.eu-central-1.amazonaws.com (s3.eu-central-1.amazonaws.com)... 185.45.7.189
Connecting to s3.eu-central-1.amazonaws.com (s3.eu-central-1.amazonaws.com)|185.45.7.189|:443... failed: Unknown error.
Retrying.

--2023-10-10 10:19:18--  (try: 2)  https://s3.eu-central-1.amazonaws.com/avg-projects/transfuser/data/14_weathers_data/Town05.zip
Connecting to s3.eu-central-1.amazonaws.com (s3.eu-central-1.amazonaws.com)|185.45.7.189|:443... failed: Unknown error.
Retrying.

--2023-10-10 10:19:41--  (try: 3)  https://s3.eu-central-1.amazonaws.com/avg-projects/transfuser/data/14_weathers_data/Town05.zip
Connecting to s3.eu-central-1.amazonaws.com (s3.eu-central-1.amazonaws.com)|185.45.7.189|:443... failed: Unknown error.
Retrying. 
```

Kait0, 2023-10-10:
People from China use VPN I think.

Lesliewsq, 2023-10-10:
I have tried VPN ，but I couldn't download the data. It seems that VPN can let me get access to those websites like ‘google.com’ or ‘youtube.com’  but  unable to use tools like  ‘wget’  or  ‘ping’.  How can I download the data by visiting certain websites instead of using ‘wget’ ？

Lesliewsq, 2023-10-10:
Thank you very much for the suggestion. It works.