autonomousvision/transfuser #194: Cannot load weights in train to finetune

MaiRajborirug, 2023-10-31:
I would like to finetune the pre-trained weight that you give us. However, there is some problem with `load_state` in `train.py`. First when I use `.load_state` the name of the state are different. For example 

Command
```terminal
$ cd team_code_transfuser
$ python train.py --batch_size 10 --load_file model_ckpt/model_2022/transfer/model_seed3_37.pth 
``` 
Result
```terminal
RuntimeError: Error(s) in loading state_dict for LidarCenterNet:
Missing key(s) in state_dict: "_model.image_encoder.features.stem.conv.weight", ...
Unexpected key(s) in state_dict: "module._model.image_encoder.features.stem.conv.weight", ...
```

After I created a function to change the state name in `.pth` I still got errors because the number of trainable parameters in the pre-trained weight and in the newly defined network is not equal. Is there anything that I can fix?

Kait0, 2023-10-31:
These models were trained with an earlier version of the codebase.
You can set `strict=False` to load the weights like it is done [during inference](https://github.com/autonomousvision/transfuser/blob/6eed5b556d318f5eff8e33a54703c23743ebdbef/team_code_transfuser/submission_agent.py#L96C21-L96C21).
You need to make sure that the error isn't actually important when using `strict=False`, since pytorch will also not complain anymore if some real problem exists.

The missing keys error depends how you train, if you train without parallization, you need to change the weight names like during [inference](https://github.com/autonomousvision/transfuser/blob/6eed5b556d318f5eff8e33a54703c23743ebdbef/team_code_transfuser/submission_agent.py#L95C20-L95C20). Using distributed parallel changes the names of the model weights.

MaiRajborirug, 2023-10-31:
Thank you for the helpful reply. However, I still have some size mismatch problem though.

Command
```python
# define model
config = GlobalConfig(root_dir='../data', setting='all') # name sensitive
device = torch.device('cuda:{}'.format(0))
print(device)
torch.cuda.set_device(device)
model = LidarCenterNet(config, device, 'transFuser', 'regnety_032', 'regnety_032', False)
model.cuda(device=device)
optimizer = optim.AdamW(model.parameters(), lr=1e-5, weight_decay=1e-4)

# load state
load_file = '/home/haoming/git/transfuser/model_ckpt/models_2022/model_seed3_37.pth'
state_dict = torch.load(load_file, map_location=device)
state_dict = {k[7:]: v for k, v in state_dict.items()}
model.load_state_dict(state_dict, strict=False)
```
Results
```terminal
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-15-244c91d2631e> in <module>
----> 1 model.load_state_dict(state_dict, strict=False)

~/anaconda3/envs/tfuse/lib/python3.7/site-packages/torch/nn/modules/module.py in load_state_dict(self, state_dict, strict)
   1603         if len(error_msgs) > 0:
   1604             raise RuntimeError('Error(s) in loading state_dict for {}:\n\t{}'.format(
-> 1605                                self.__class__.__name__, "\n\t".join(error_msgs)))
   1606         return _IncompatibleKeys(missing_keys, unexpected_keys)
   1607 

RuntimeError: Error(s) in loading state_dict for LidarCenterNet:
	size mismatch for _model.lidar_encoder._model.conv1.weight: copying a param with shape torch.Size([32, 3, 3, 3]) from checkpoint, the shape in current model is torch.Size([32, 2, 3, 3]).
```

Kait0, 2023-10-31:
You probably need to overwrite global config values with the configuration values from the args.txt as is done [here](https://github.com/autonomousvision/transfuser/blob/6eed5b556d318f5eff8e33a54703c23743ebdbef/team_code_transfuser/submission_agent.py#L54C10-L54C10).

MaiRajborirug, 2023-11-01:
It is working now when I load state the way of  `submission_agent.py`. Thank you so much for your timely response!

closed 2023-11-01 by Kait0

MaiRajborirug, 2023-11-01:
I wonder do you have the `optimizer.pth` for the pretrained weight? If not, what should be the start learning rate if I want to finetune the model?

```python
optimizer = optim.AdamW(model.parameters(), lr=10e-4) # original learning rate 
```

Kait0, 2023-11-02:
I don't have the optimizer weights anymore. You can either use the start learning rate or the final learning rate which is 10x smaller. (But you will need to tune the learning rate anyway for your task).