autonomousvision/transfuser #196: Finetune without `optimizer.pth`

MaiRajborirug, 2023-11-14:
As a followup from [ISSUE 194](https://github.com/autonomousvision/transfuser/issues/194) The given pre-train weights don't have `optimizer.pth` along with it. However, when we load weight states. We are supposed to load an optimizer as shown in [train.py](https://github.com/autonomousvision/transfuser/blob/2022/team_code_transfuser/train.py) line 183. Since we didn't have `optimizer.pth`. from the pre-trained folder What should I do to make the training go smoothly?

Kait0, 2023-11-14:
Just comment that line. 
I would expect that supervised finetuning works if you start with fresh adam momentums.
Similar than starting a new training with pre-trained image net weights (where you also don't load the optimizer weights).

MaiRajborirug, 2023-11-15:
Got it thank you so much!

MaiRajborirug, 2023-11-15:
Then what should be the learning rate? Should I keep learning rate 10e-4 or should I take into account that the model already trained for 41 epoch and reduce the start learning rate in the model?

Kait0, 2023-11-16:
Well its not possible to predict the appropriate learning rate for a task in advance. Start with the default in the repository and test different values.

closed 2023-11-21 by MaiRajborirug