autonomousvision/transfuser #198: The problem of reconstructing the dataset

carryman97, 2023-12-04:
Hello, I tried to create a dataset, but I encountered a problem. I don't know how to reconstruct the two dataset files "right_dataset" and "left_dataset". When I did not use these two dataset files to train the model agent, I found that the agent turned left or right at the intersection very poorly and could not correctly drive into the prescribed lane. I hope you can help me solve the problem. What should I do to rebuild the two dataset files "right_dataset" and "left_dataset"? Thank you!

carryman97, 2023-12-05:
> See #188.
>  However TransFuser should be able to make turns consistently even without these routes so I suspect your problem lies elsewhere.

I reconstructed the dataset based on your open source data collection code, and when I reconstructed the dataset, each route was randomly assigned a type of weather (sunny, cloudy, rainy, etc.), light level, sun azimuth, etc., instead of Randomly assign these parameters to each frame (because I want to change the model input from a single frame input to a time series frame input. If the scenes of adjacent frames are different, then there will be no correlation between the preceding and following frames. For this I Need to rebuild the dataset). When I collected the data and used it to train the same model, I got bad results. I want to ask you, did you clean the data after collecting the data? For example, delete data where collisions occurred, or delete data where the route was not completed. Or do you have any other good suggestions? The problem of data sets has troubled me for a long time.Thank you！

crazyman
***@***.***

Kait0, 2023-12-05:
Bad result is too vague to give you any concrete advise.
We didn't do any particular clearning in this repository.
I would first make sure your training pipeline is working properly. Train with the original dataset and see if you can reproduce the result. If not than there is some problem with the training setup.

Also I wanted to mention, in our latest repository for [TransFuser++](https://github.com/autonomousvision/carla_garage) we use a dataset that changes the weather etc. in every route (except every frame), like you need.

closed 2023-12-11 by Kait0