autonomousvision/transfuser #201: what is the range of the BEV map？

BestBUCKY, 2024-01-29:
Dear author， thanks for your work！
I am curious about the topdown files. I notice the vehicle in the map is 30m around，but what is the range of drivable area and line？ and how can change the map range？

kashyap7x, 2024-02-05:
The topdown files are 500x500 pixels at a resolution of 5 pixels per meter (so totally 100m x 100m with the ego vehicle at the center).

You can find the relevant code here: https://github.com/autonomousvision/transfuser/blob/22b3ccdb7e806b42c9be2aa476b71546e8ec3620/team_code_autopilot/utils/lts_rendering.py#L16

BestBUCKY, 2024-02-06:
> The topdown files are 500x500 pixels at a resolution of 5 pixels per meter (so totally 100m x 100m with the ego vehicle at the center).
>
> You can find the relevant code here: [...]
Thanks for your reply!! I understrand

closed 2024-02-06 by BestBUCKY