autonomousvision/transfuser #203: Need some help about understanding the code

Oliverwang11, 2024-03-07:
Hi, 

Thanks again for your contribution!

After read the paper, I took a look at the code, especially for the GPT class, but I found something I am a little bit confused. 

1. In the paper, it says the the input image is down-sampled to 5*22xC and LiDAR to 8*8*C. If I understand correctly, for inference batch size, in your comments B, should be one? And why the input size of image is B*4*seq_len, C, H, W in your comments, where does the number 4 come from? Maybe I misunderstood some thing.

`    def forward(self, image_tensor, lidar_tensor, velocity):
        """
        Args:
            image_tensor (tensor): B*4*seq_len, C, H, W
            lidar_tensor (tensor): B*seq_len, C, H, W
            velocity (tensor): ego-velocity
        """
`
3. BTW can I skip using Carla, since what I am interesting in is the matrices size of the intermediate features, is it possible to run it with some fake input?

Best wishes!
Thanks again!

Kait0, 2024-03-08:
I think it's just a typo.

Oliverwang11, 2024-03-08:
Thanks for your reply, so the batch size, B, during inference is one right?

Oliverwang11, 2024-03-08:
Thanks a lot!

closed 2024-03-08 by Oliverwang11