autonomousvision/transfuser #205: Some issues about vehicle speed and data collection.

Kun-k, 2024-03-26:
Sorry to bother you, but I've encountered some issues regarding vehicle speed and data collection. 

I've selected a section of Town04 and aim to control vehicles using the algorithms and models provided by you. I've noticed that in the absence of any surrounding vehicles, the autonomous vehicle can only drive at a speed of 3.89. This speed is quite slow. Could there be an issue with my scenario or experimental setup, or does the pretrained model support only lower speeds?

The second issue pertains to data collection. I intend to gather data on a section of the Town04 to train a model capable of driving at higher speeds. I need to remove all traffic lights from this section of the route. Is this possible?

Kait0, 2024-03-26:
There is no issue here, the model is trained to only drive at speeds up to 4 m/s = 14.4 km/h.
This is a result of the rule based expert driver that collected the data only driving at that speed at most.

We improved this in our new model [TransFuser++](https://github.com/autonomousvision/carla_garage) which can drive at speeds up to 8 m/s = 29 km/h, (although the best performance is attained at 6 m/s).
As far as I know there is no model in the CARLA leaderboard 1.0 that drives faster than ~30 km/h, because the background traffic is only driving at that speed.

The CARLA team improved this in the CARLA leaderboard 2.0 and new simulator versions (>= CARLA 0.9.14) where I think the background traffic can drive up to ~100 km/h, and agents are penalized in the DS if they drive far below the speed limit. Unfortunately, there are no publicly available pre-trained models for leaderboard 2.0 at this time, as far as I know.

I have never tried to remove traffic lights. Traffic lights are of the class [carla.Actor](https://carla.readthedocs.io/en/latest/python_api/#carlaactor) which does have a .destroy() method.
You could try something like this, though I don't know what will happen exactly.

```
    all_actors = world.get_actors()
    all_traffic_lights = all_actors.filter('*traffic_light*')
    for actor in all_traffic_lights:
         actor.destroy()
```

Kun-k, 2024-03-27:
Thank you for your reply. I believe this can solve my problem. Also, I really appreciate your great work.

closed 2024-03-27 by Kun-k