autonomousvision/transfuser #206: Potential bug in the value of i_stuck?

YuhengZhi, 2024-03-31:
https://github.com/autonomousvision/transfuser/blob/22b3ccdb7e806b42c9be2aa476b71546e8ec3620/team_code_autopilot/autopilot.py#L726-L745

Should this line `i_stuck = i` be put in the following for-loop? like:
```Python
for id, traffic_participant in nearby_vehicles.items():
    i_stuck = i
    if self.render_bev==False and self.junction==False and i > number_of_future_frames_no_junction:
            break
    if id in tmp_stucked_vehicle_id:
        i_stuck = 0
```

If not, can someone please explain the reason to change the number of future time frame being checked to 0 for *all remaining* vehicles once *any* vehicle is found stuck?

Kait0, 2024-04-03:
Seems like a bug yes.
You can check if it makes a difference to the DS of the expert.

Kait0, 2024-06-28:
Came around to test it. Didn't seem to have a significant impact on performance. I uploaded the fix.

closed 2024-06-28 by Kait0