autonomousvision/transfuser #208: Dependencies error to train the transfuser model

SRajasekar333, 2024-04-08:
Hi,

As discussed in https://github.com/autonomousvision/transfuser/issues/114#issue-1406818291, I installed the necessary dependencies for the conda environment. But when I train the model, I face the following error, please help to resolve and guide me to train the model.  
[image]
Thanks in advance.

Kait0, 2024-04-09:
You can just comment [this line](https://github.com/autonomousvision/transfuser/blob/22b3ccdb7e806b42c9be2aa476b71546e8ec3620/team_code_transfuser/model.py#L11C26-L11C40).
Its just for an ablation (you likely don't need it) but unfortunately introduces tricky cuda dependencies.

SRajasekar333, 2024-04-11:
Hi,
Thanks for the reply.
Yes I corrected the code and trained the model, but now I face error in Multiprocessing.
[image]
Could you please help me, due to what reason I face this error and how to troubleshoot it?
Thanks in advance.

Kait0, 2024-04-11:
This might not actually be an error with multiprocessing. The start method of... message is normal info printing.
You need to debug line by line, either with prints or a debugger to see where exactly the code crashes / gets killed by the os.

closed 2024-04-18 by Kait0