autonomousvision/transfuser #212: CARLA setup error and Running CARLA server without display for evaluation

SRajasekar333, 2024-05-06:
Hi,
I am facing some doubts while trying to evaluate the model, so please guide. 
I trained the transfuser model by running 
`CUDA_VISIBLE_DEVICES=0,1,2,3,4,5,6 OMP_NUM_THREADS=48 OPENBLAS_NUM_THREADS=1 torchrun --nnodes=1 --nproc_per_node=7 --max_restarts=0 --rdzv_id=1234576890 --rdzv_backend=c10d train.py --logdir log --root_dir data --parallel_training 1 --epochs 32 --batch_size 48 --id transfuser` 
and saved the 32 `model_n.pth` and `optimizer_n.pth`

So now I'm trying to evaluate the model as in `README.md`, 

1. When I setup and install the CARLA 0.9.10.1, I face the following error `ERROR 404: Not Found`

[image]

2. I'm trying to evaluate the model using the HPC remote server, so in this case how to run the CARLA server without display?

    Can I run the server for **Without Display - Without Docker** condition as,
    `SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh --world-port=2000 -opengl`
     and then evaluate the agent using,
     `./leaderboard/scripts/local_evaluation.sh <carla root> <working directory of this repo (*/transfuser/)>`

    Please guide on how to evaluate the model and save the output files instead of stream display.

Thanks in advance.

Regards,

Kait0, 2024-05-06:
1. The download link for carla changed. You can use [this script](https://github.com/autonomousvision/carla_garage/blob/main/setup_carla.sh) to download it (will update later).

2. If the carla server and the evaluation script are on the same machine then yes, otherwise you need to set the ip and port etc.

Kait0, 2024-05-14:
I updated the links in the download scripts.

closed 2024-05-14 by Kait0