autonomousvision/transfuser #215: I met an Error during evaluation. How can I solve it?

jingzhengli, 2024-05-21:
========= Preparing RouteScenario_33 (repetition 0) =========
> Setting up the agent

Could not set up the required agent:
> module 'transfuser' has no attribute 'get_entry_point'

Traceback (most recent call last):
  File "/home/ljzdisk/ad/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 272, in _load_and_run_scenario
    agent_class_name = getattr(self.module_agent, 'get_entry_point')()
AttributeError: module 'transfuser' has no attribute 'get_entry_point'
> Registering the route statistics

jingzhengli, 2024-05-21:
> ========= Preparing RouteScenario_33 (repetition 0) ========= _>_ Setting up the agent
>
> Could not set up the required agent: _>_ module 'transfuser' has no attribute 'get_entry_point'
>
> Traceback [...]

It is not clearly how to set "export TEAM_AGENT=${WORK_DIR}/team_code_transfuser/**submission_agent.py**" in local_evaluation.sh
I set to "export TEAM_AGENT=${WORK_DIR}/team_code_transfuser/**transfuser.py"**

jingzhengli, 2024-05-21:
> It is not clearly how to set "export TEAM_AGENT=${WORK_DIR}/team_code_transfuser/**submission_agent.py**" in local_evaluation.sh I set to "export [...]

Then, It doesn't have any error, but report this imformation.

**_>_ Registering the global statistics**

Kait0, 2024-05-21:
what is not clear? You don't need to change this. submission_agent.py is the correct file.
The transfuser.py file only contains the transfuser backbone, there are other files involved.

> Registering the global statistics

This can happen if your code crashed before or so. Try deleting the ${WORK_DIR}/results/transfuser_longest6.json file and try again (might reveal the real error).

jingzhengli, 2024-05-21:
> > what is not clear? You don't need to change this. submission_agent.py is the correct file. The transfuser.py file only contains the transfuser backbone, there are other files involved.
> > > Registering the global statistics
> > 
> > 
> > This can happen if your code crashed before or so. Try deleting the ${WORK_DIR}/results/transfuser_longest6.json file and try again (might reveal the real error).
> 
> Thanks for your reply. If I use this comment: "export TEAM_AGENT=${WORK_DIR}/team_code_transfuser/submission_agent.py". I will met an error like this： **./leaderboard/scripts/local_evaluation.sh /home/ljzdisk/carla /home/ljzdisk/ad/transfuser ./leaderboard/scripts/local_evaluation.sh: line 38: 69613 Segmentation fault (core dumped) python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator_local.py --scenarios=${SCENARIOS} --routes=${ROUTES} --repetitions=${REPETITIONS} --track=${CHALLENGE_TRACK_CODENAME} --checkpoint=${CHECKPOINT_ENDPOINT} --agent=${TEAM_AGENT} --agent-config=${TEAM_CONFIG} --debug=${DEBUG_CHALLENGE} --resume=${RESUME}** I cannot know where is the error. Thanks again. I want to know if the error come from the version of carla==0.9.15 that I installed early.

Now, I know the error comes form **"self.module_agent = importlib.import_module(module_name)" in leaderboard_evaluator_local.py**. But I don't know why it happens and how to solve it. Thanks again,

jingzhengli, 2024-05-21:
> Now, I know the error comes form **"self.module_agent = importlib.import_module(module_name)" in leaderboard_evaluator_local.py**. But I don't know why it happens and how to solve it. Thanks again,

The same error as previous issue "https://github.com/autonomousvision/transfuser/issues/200".

jingzhengli, 2024-05-21:
> The same error as previous issue "#200".

> This codebase is not compatible with carla 0.9.15 (You need leaderboard 2.0 for that and as far as I know there are no public codebases available for that yet). You need carla version 0.9.10.1 to run the code. The [setup script](https://github.com/autonomousvision/transfuser/blob/2022/setup_carla.sh) will download it for you.

Many thanks.

closed 2024-05-21 by jingzhengli