autonomousvision/transfuser #218: error while data collection

SExpert12, 2024-05-24:
Hi, 
I want to create dataset and for that I run like this:
./datagen.sh /home/carla /home/transfuser-2022

But I got this error.

 can't open file '/home/transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py': [Errno 2] No such file or directory

 I have check my folders, file is there but it is still showing this error. How to solve this?

~/transfuser-2022/leaderboard/leaderboard$ ls
autoagents  envs  __init__.py  leaderboard_evaluator_local.py  leaderboard_evaluator.py

Thanks.

SExpert12, 2024-05-25:
No, It is actually given /home/username/transfuser-2022. I deleted it.

Kait0, 2024-05-28:
No CARLA 0.9.14 is not compatible with this codebase/any codebase using leaderboard 1.0.
You need to use CARLA version 0.9.10.1
[This script](https://github.com/autonomousvision/transfuser/blob/2022/setup_carla.sh) will download the correct version for you.

To use CARLA 0.9.14 you need to use leaderboard 2.0, but there are no publicly available repositories for it right now.
(You can adapt existing leaderboard 1.0 codebases to become compatible with 2.0, it will require some work though).

SExpert12, 2024-05-29:
Thanks for your reply and suggestion.

closed 2024-05-29 by Kait0