autonomousvision/transfuser #220: FOV in the data

st3lzer, 2024-05-28:
Hello,
I have question regarding the FOV of the data. 
For training data generation, three cameras with a FOV of 60° each are used. The Images are concatinated and cropped to 704x160. 

In inference, we have three cameras with a FOV of 120° which are cropped to a third in each dimension. This is done because there are distortions in the cameras for inference (#171). But wouldn’t the cropping to a third result in three images with a horizontal FOV of 40° each, which are then concatinated? Even if there are no distortions in the camera in inference, I tested and the images fit perfectly together. 

Thank you!

Kait0, 2024-05-28:
> But wouldn’t the cropping to a third result in three images with a horizontal FOV of 40° each, which are then concatinated?

Nope, cropping the center third of a 120° camera yields a 60° FoV (try drawing it with a pinhole camera model).
Degrees are not evenly distributed along the image.

st3lzer, 2024-05-28:
Thank you for the very fast reply, this makes sense!

Is there a reason why the distorted cameras are used in the inference?

Kait0, 2024-05-28:
Because it is mandatory on the leaderboard 1.0 test server to use the distorted cameras.

closed 2024-05-28 by st3lzer