autonomousvision/transfuser #222: 'AttributeError' and 'UnboundLocalError' when running local_evaluation.sh

rogerfinnerty, 2024-05-30:
Hello, 

I'm trying to run the local evaluation script and am getting the following errors:

[image]

I've set up the environment and am running a Carla instance, I've also changed the paths in the shell script ... how can I fix this?

Thanks!

Kait0, 2024-05-31:
You need to send the entire stacktrace. This is a secondary error. The real problem likely came earlier.

rogerfinnerty, 2024-05-31:
Apologies, here is the rest of the stack trace
[image]

Kait0, 2024-06-03:
That means the torch_scatter module didn't install correctly.
The easy workaround is to just comment out [this line](https://github.com/autonomousvision/transfuser/blob/b602efac325e77f853460ba0d79dc364c2c70488/team_code_transfuser/model.py#L11), you will likely not need it.

closed 2024-06-06 by Kait0