autonomousvision/transfuser #224: The training effect is very poor！

zygalaxy, 2024-06-17:
I used four 3080ti, the CPU was 40 cores, and I trained for 41 times, the training duration was 37 hours, and the training code was CUDA_VISIBLE_DEVICES=0,1,2. 3 OMP_NUM_THREADS=40 OPENBLAS_NUM_THREADS=1 torchrun --nnodes=1 --nproc_per_node=4 --max_restarts=0 --rdzv_id=1234576890 --rdzv_backend=c10d train.py - -root _ dir/home/Ubuntu/zy/zy _ 2t/project _ auto/transfuser/dataset-parallel _ training1-batch _ size12, not using all 210G data sets, but only using about 160G data sets, as shown in the following figure.
[image]
However, the test result of the trained model is very poor. The test code is, and the visual results are as follows bash leaderboard/scripts/local_evaluation.sh. 
[image]
What is the problem? I hope to get a solution, and I will be very grateful.

Kait0, 2024-06-18:
This looks like a bug or something is wrong with the setup.
Could you set loading to [strict](https://github.com/autonomousvision/transfuser/blob/b602efac325e77f853460ba0d79dc364c2c70488/team_code_transfuser/submission_agent.py#L96) to see whether the model was loaded correctly.
Did the training losses go down properly?

zygalaxy, 2024-06-19:
I followed your instructions and encountered two issues:
1)My total training loss is shown in the figure below.
[image]
2)The error message when strict is set to True is different for my model and the Transfuser model (both fail to execute with strict set to True). Figure 1 shows the error message for my_train_model, and Figure 2 shows the error message for the Transfuser model (since the full message cannot be displayed, I have also submitted error.txt for my_model).
[image]
[image]
[error.txt](https://github.com/user-attachments/files/15894337/error.txt)

The loss seems normal, so as you mentioned, the issue is likely related to model loading. However, based on the error messages, I can't determine the exact problem. I hope to get your advice, thank you!

Kait0, 2024-06-19:
The error for the Transfuser model is normal, it was trained with an older codebase.

It probably has something to do with the name.
Did you change how the model was saved [here](https://github.com/autonomousvision/transfuser/blob/b602efac325e77f853460ba0d79dc364c2c70488/team_code_transfuser/train.py#L383)?

When trained with distributed data parallel we remove part of the name that comes from ddp [here](https://github.com/autonomousvision/transfuser/blob/b602efac325e77f853460ba0d79dc364c2c70488/team_code_transfuser/submission_agent.py#L95).

Are there any optimizer files in your model folder (that could get loaded accidentally)?

zygalaxy, 2024-06-19:
You are right. I found that my training results seem to contain optimization files (as shown in Figure 1 below). My submission_agent.py is shown in Figure 2 below, and the training code is CUDA_VISIBLE_DEVICES=0,1,2. 3 OMP_NUM_THREADS=40 OPENBLAS_NUM_THREADS=1 torchrun --nnodes=1 --nproc_per_node=4 --max_restarts=0 --rdzv_id=1234576890 --rdzv_backend=c10d train.py - -root _ dir/home/Ubuntu/zy/zy _ 2t/project _ auto/transfuser/dataset-parallel _ training1-batch _ size12, what's wrong with this training result? When I use model_41.pth+optimizer_41.pth, the problem is that the vehicle directly turns right and deviates from the route. When I only use the model_41.pth file, the problem is that the vehicle does not move at all (if the transfuser model is used, the vehicle runs well).
[image]
[image]
[image]

Kait0, 2024-06-19:
Your training result seems correct.
Inference needs to be different. The correct way to use the code is to copy the args.txt + model_31.pth file into a new folder (no optimizer files) and use this folder for inference.
Then you should not get any error when you load the model with strict.
Can you send the visualization file again when you do this.

Kait0, 2024-06-19:
Typically epoch 31 yields the best or close to the best results. 
Sometimes TransFuser gets stuck due to causal confusion. If it does not happen very often this is normal.
After a while a crepping mechanism will unblock it if this happens. This is discussed in Table 10 of the paper.

zygalaxy, 2024-06-20:
Then it means that I should directly change the default epoch=41 to 31 and use 31 as the final model, which can also reduce the training time.

Kait0, 2024-06-20:
you can do that. We have done that as well in our follow up work.

zygalaxy, 2024-06-20:
Ok, thank you for your patience. I appreciate it very much. I will close this question.

closed 2024-06-20 by zygalaxy