autonomousvision/transfuser #227: Accessing the gradient of the transfuser model in evaluation mode

MCUBE-2023, 2024-07-03:
Hi, 

Would please tell me which part of the code allows me to access the gradient of the transfuser model in evaluation mode ?

Thank you bunches in advance :)

Kait0, 2024-07-03:
If you mean gradients during deployment, than you need to look at the [run_step](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L236) function.
Gradient computation is disabled by default to improve efficiency, you need to remove the inference_mode() and no_grad() decorators.

MCUBE-2023, 2024-07-06:
Thank you so much for your quick reply. Your help allowed me to resolve an issue in which I've blocked for more then one month. I acknowledge your highly developed professional qualities :)

closed 2024-07-07 by Kait0