autonomousvision/transfuser #228: How to Change vehicle model for evaluation

tijaz17skane, 2024-07-12:
I want to change the default vehicle model that gets loaded upon running ./local_evaluation.sh to carlamotors.firetruck. I tried changing vehicles.* on line 184 in ./leaderboard_evaluator_local.py to vehivle.carlamotors.firetruck, but that didn't help.

Kait0, 2024-07-12:
The vehicle model type is specified [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/leaderboard/scenarios/route_scenario_local.py#L250)

tijaz17skane, 2024-07-12:
> The vehicle model type is specified [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/leaderboard/scenarios/route_scenario_local.py#L250)

I did that. changed it to carlamotors.firetruck, but it didn't help

Kait0, 2024-07-12:
hm if you are for some reason not useing the route_scenario_local file, you need to edit [route_scenario.py](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/leaderboard/scenarios/route_scenario.py#L249)
Or if you are using scenario runner (probably not) the respective function is [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/scenario_runner/srunner/scenarios/route_scenario.py#L213)

tijaz17skane, 2024-07-15:
I edited all three files one by one. and changed all lincoln cars to firetruck but it makes no difference, i'm getting the same car even after all those changes.

Kait0, 2024-07-15:
Hm perhaps the issue is that the carlamotors.firetruck is not available in CARLA 0.9.10 (Was introduced in CARLA 0.9.12, but I think this codebase is not compatible with CARLA 0.9.12).
You should try some of the available vehicles (https://carla.readthedocs.io/en/0.9.10/bp_library/#vehicle).
If you still see the lincoln.mkz2017 with a valid vehicle type then the code probably is not executed (maybe there is a pythonpath pointing to the leaderboard code of another repository?) You need to debug then to see that the code is actually executed.

closed 2024-07-17 by tijaz17skane