autonomousvision/transfuser #230: Attention Map Visualizations- Obsolete version

MCUBE-2023, 2024-07-14:
Some people asked in the closed issues about Attention Map Visualizations (such as #18), but in #18 you pointed out that we should use viz.py to make these visualizations, which I think it is related to an obsolete version of the Transfuser model (CVPR 2021), and please correct if I am wrong. 

Would you please point out (or refer me) to the adequate part of the code, so I can visualize the attention maps of the transfuser model (not the CVPR 2021 version) ?

ap229997, 2024-07-22:
Check [viz.py](https://drive.google.com/file/d/1CPI2BxvG0XC3d9DcWEl565aZqV9Mz3Q9/view?usp=sharing), this also requires the attention map to be returned from the transformer backbone as shown [here](https://drive.google.com/file/d/1FwQlwcot09K0MTEhMXeDEgr6jZRmg-Q7/view?usp=sharing). Note that this code may not be compatible with the latest code in this repo, but this should give enough context on how to visualize attention maps.

Mr-ChenSH, 2024-07-29:
May I ask if there is a download link for the dataset? The script may not be able to download

closed 2024-08-04 by Kait0

MCUBE-2023, 2024-08-04:
@ap229997 I am currently working on adapting the viz.py (https://drive.google.com/file/d/1CPI2BxvG0XC3d9DcWEl565aZqV9Mz3Q9/view) that you shared with me in this issue to suit the transfuser version of 2022. I succeeded to modify to transfuser.py (2022) to return the attention maps. However, I am finding difficulties to adapt viz.py to suit the transfuser version of 2022, so I can visualize the tokens of the attention maps.

When adapting viz.py to the transfuser version of 2022, is it possible to visualize the attended tokens related to the attention maps in 'evaluation mode' without requiring the training data? For example, I see that in viz.py (https://drive.google.com/file/d/1CPI2BxvG0XC3d9DcWEl565aZqV9Mz3Q9/view), there are these 2 lines of code which are related to data: 
		rgb = data['rgb'].to(args.device, dtype=torch.float32)
		lidar = data['lidar'].to(args.device, dtype=torch.float32)
		
Thanks in advance !

MCUBE-2023, 2024-08-07:
@Kait0  I will be really grateful if you can help me find an answer for the last question that I asked in this issue.

Kait0, 2024-08-07:
I have not used this file, but I don't see why not. Just replace rgb and lidar with the rgb and lidar you get in evaluation mode. Perhaps you need to make sure that the preprocessing is the same.

MCUBE-2023, 2024-08-14:
@Kait0 Thank you for your quick reply :)